ChristianTremblay/BAC0 #556: Cannot Read Proprietary Object in Async BAC0 (2025.6.10)

likair, 2025-08-07:
# Test 1

```
await bacnet.read("2001:2 129 1 513")
``` 

Output
```
-no object class-
```

> This test works with the legacy `bac0@23.99` with the correct response.

# Test 2

Based on https://bac0.readthedocs.io/en/stable/proprietary_objects.html#

```
await bacnet.read("2001:2 @obj_129 1 @prop_513")
```

Output
```
raise ValueError("'type,instance' or 'type:instance' expected")
```

ChristianTremblay, 2025-08-11:
bacpypes3 works differently and we need to explicitly define classes to specify the type of the property.

For example, here is an example for a JCI device trying to read proprietary property 603 of the device object
```
await bacnet.read('303:9 device 5209 @prop_603') 
'-no property type-'
>>> await bacnet.read('303:9 device 5209 603')       
'-no property type-'
>>> from BAC0.core.proprietary_objects import jci_5 
>>> await bacnet.read('303:9 device 5209 603')      
0
```

So, as you can see, until I import the special class defining proprietary objects for jci, bacpypes doesn't know what to do.

Can you share what controller you are trying to read ?

TomasPaier, 2025-09-30:
Hello likair,
could you please post / send me your functional proprietary object definition file (as an example). I'm trying to create a functional version of Produal under the new BAC0 but with no success:
Exception has occurred: AttributeError
not a property identifier: PROP_40035...
Thank you!

TomasPaier, 2025-09-30:
[produal_651.py](https://github.com/user-attachments/files/22623965/produal_651.py)
This is my file...

ChristianTremblay, 2025-10-01:
Haaaaa ! This explains why I didn't have issues on my side (mostly working with JCI controllers here)

It is wonderful to see you guys working together and making it work. Thanks for providing feedback.